Duet3D/DuetSoftwareFramework #112: DCS Pre-1.3.2 Cancel while pause is still executing is ignored

gtjoseph, 2020-04-11:
If you Cancel while Pause is still running, you get "Error: Code has been cancelled" and the Cancel itself is ignored.

```
Apr 11 15:03:04 duet3j.f5.int DuetControlServer[4264]: [debug] Completed G1 X43.073 Y-88.789 E0.0798 F3263 =>
Apr 11 15:03:04 duet3j.f5.int DuetControlServer[4264]: [debug] Waiting for execution of G1 X41.777 Y-129.906 E2.7989
Apr 11 15:03:04 duet3j.f5.int DuetControlServer[4264]: [debug] Processing G1 X41.777 Y-129.906 E2.7989
Apr 11 15:03:04 duet3j.f5.int DuetControlServer[4264]: [debug] Completed G1 X44.311 Y-89.082 E0.1591 =>
Apr 11 15:03:04 duet3j.f5.int DuetControlServer[4264]: [debug] Waiting for execution of G1 X40.911 Y-129.906 E2.8529
Apr 11 15:03:04 duet3j.f5.int DuetControlServer[4264]: [debug] Processing G1 X40.911 Y-129.906 E2.8529
Apr 11 15:03:04 duet3j.f5.int DuetControlServer[4264]: [debug] File: Sent G1 X41.777 Y-129.906 E2.7989, remaining space 224, needed 48
Apr 11 15:03:04 duet3j.f5.int DuetControlServer[4264]: [debug] Completed G1 X44.311 Y-130.137 E2.7194 =>
Apr 11 15:03:04 duet3j.f5.int DuetControlServer[4264]: [debug] Completed G1 X43.070 Y-130.421 E2.7988 =>
Apr 11 15:03:04 duet3j.f5.int DuetControlServer[4264]: [debug] Waiting for execution of G1 X40.911 Y-89.306 E5.3848
Apr 11 15:03:04 duet3j.f5.int DuetControlServer[4264]: [debug] Waiting for execution of ; feature internal single extrusion
Apr 11 15:03:04 duet3j.f5.int DuetControlServer[4264]: [debug] Processing G1 X40.911 Y-89.306 E5.3848
Apr 11 15:03:04 duet3j.f5.int DuetControlServer[4264]: [debug] File: Sent G1 X40.911 Y-129.906 E2.8529, remaining space 176, needed 48
Apr 11 15:03:04 duet3j.f5.int DuetControlServer[4264]: [debug] File: Sent G1 X40.911 Y-89.306 E5.3848, remaining space 128, needed 48
Apr 11 15:03:04 duet3j.f5.int DuetControlServer[4264]: [debug] Processing ; feature internal single extrusion
Apr 11 15:03:04 duet3j.f5.int DuetControlServer[4264]: [debug] Completed ; feature internal single extrusion
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] IPC#111: Got new UNIX connection, checking mode...
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] IPC#111: Command processor added
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] IPC#111: Received command SimpleCode
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Waiting for execution of M25
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Processing M25
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] HTTP: Sent M25, remaining space 1512, needed 24
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X41.797 Y-130.506 E2.8784
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X9.237 Y-130.506 E4.9089
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X7.962 Y-130.421 E4.9886
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X40.911 Y-89.306 E5.3848
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X6.713 Y-130.131 E5.0686
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X6.713 Y-89.084 E7.6284
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X7.961 Y-88.789 E7.7084
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X9.238 Y-88.706 E7.7881
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X40.911 Y-129.906 E2.8529
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X41.796 Y-88.706 E9.8186
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G92 E0.0000
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X35.313 Y-129.906 E2.5319 F2025
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X32.513 Y-129.906 E2.7065
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X32.513 Y-89.306 E5.2385
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X41.777 Y-129.906 E2.7989
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X35.313 Y-89.306 E5.4131
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X42.983 Y-129.826 E2.7235
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X43.711 Y-129.659 E2.6769
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X43.711 Y-89.557 E0.1760
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X35.313 Y-89.306 F4800
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X36.713 Y-89.306 F4800
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X42.984 Y-89.384 E0.1294
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X41.777 Y-89.306 E0.0540 F2025
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G92 E0.0000
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X39.511 Y-89.306 E0.1745 F2025
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X40.911 Y-89.306 F4800
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X36.713 Y-129.906 E2.8809
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X36.713 Y-89.306 E5.4128
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled G1 X39.511 Y-129.906 E2.7064
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [info] Job has been paused at byte 65514, reason User
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [info] Starting macro file pause.g on channel HTTP
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Waiting for execution of ; pause.g (macro code)
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Waiting for execution of ; called when a print from SD card is paused (macro code)
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Processing ; pause.g
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Completed ; pause.g
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Processing ; called when a print from SD card is paused
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Waiting for execution of M83 ; relative extruder moves (macro code)
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Completed ; called when a print from SD card is paused
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Waiting for execution of G91 ; relative positioning (macro code)
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Processing M83 ; relative extruder moves
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Waiting for execution of G1 X5 Y50 F3600 (macro code)
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Processing G91 ; relative positioning
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Waiting for execution of G1 Z50 E-5 F600 (macro code)
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Waiting for execution of G90 (macro code)
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Processing G1 X5 Y50 F3600
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Processing G1 Z50 E-5 F600
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Processing G90
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] HTTP: Sent M83 ; relative extruder moves, remaining space 1512, needed 24
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] HTTP: Sent G91 ; relative positioning, remaining space 1488, needed 24
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] HTTP: Sent G1 X5 Y50 F3600, remaining space 1440, needed 48
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] HTTP: Sent G1 Z50 E-5 F600, remaining space 1392, needed 48
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] HTTP: Sent G90, remaining space 1368, needed 24
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Completed M83 ; relative extruder moves
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Completed G91 ; relative positioning
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Completed G1 X5 Y50 F3600
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Completed G1 Z50 E-5 F600
Apr 11 15:03:05 duet3j.f5.int DuetControlServer[4264]: [debug] Completed G90
[... cut ...]
Apr 11 15:03:11 duet3j.f5.int DuetControlServer[4264]: [debug] Completed M25 => Printing paused at X6.1 Y-130.6 Z0.2
Apr 11 15:03:11 duet3j.f5.int DuetControlServer[4264]: [debug] HTTP: ==> Starting code: M25 => Printing paused at X6.1 Y-130.6 Z0.2
Apr 11 15:03:11 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled M0 H1
Apr 11 15:03:11 duet3j.f5.int DuetControlServer[4264]: [debug] IPC#111: Connection closed
Apr 11 15:03:11 duet3j.f5.int DuetControlServer[4264]: [debug] IPC#112: Connection closed
```

chrishamm, 2020-04-14:
Thanks, I could reproduce this problem and it should be fixed in the latest commit.

gtjoseph, 2020-04-14:
Not fixed.   Pressing cancel which the pause macro is still executing simply prints "Error: Code has been cancelled".

chrishamm, 2020-04-14:
I cannot reproduce this error message but I did notice that M0 could be processed before M25 finishes. I'll implement a work-around.

gtjoseph, 2020-04-14:
The "Error: " but might be my prepending the text to the message but the "Code has been cancelled" is coming from the DCS and does _not_ show in the DCS debug log.

If you look at the last few lines of the debug log I pasted above, you'll see that indeed, the M0 is being processed before the M25 completes.

```
Apr 11 15:03:07 duet3j.f5.int DuetControlServer[4264]: [debug] Processing M0 H1
Apr 11 15:03:11 duet3j.f5.int DuetControlServer[4264]: [info] HTTP: Finished macro file pause.g
Apr 11 15:03:11 duet3j.f5.int DuetControlServer[4264]: [debug] Completed M25 => Printing paused at X6.1 Y-130.6 Z0.2
Apr 11 15:03:11 duet3j.f5.int DuetControlServer[4264]: [debug] HTTP: ==> Starting code: M25 => Printing paused at X6.1 Y-130.6 Z0.2
Apr 11 15:03:11 duet3j.f5.int DuetControlServer[4264]: [debug] Cancelled M0 H1
```

chrishamm, 2020-04-15:
I just pushed another commit which is likely to fix this problem and possibly the other one. Please test it when you get the chance.

gtjoseph, 2020-04-16:
Confirmed Fixed

closed 2020-04-16 by gtjoseph