Duet3D/DuetSoftwareFramework #113: DCS Pre-1.3.2 Pause/Resume seems to be dropping moves left in buffer

gtjoseph, 2020-04-12:
See the picture from https://forum.duet3d.com/post/145573

The extruder was drawing a brim counter-clockwise and the pause was done at the 7 o'clock position.  My pause script moves the extruder without an increase in Z to prevent a blob from forming, then raises Z.  When I resumed, the extruder returned to the spot it was paused but instead of continuing to follow the outline, the extruder moved directly  to the 3 o'clock position and continued drawing the outline counter-clockwise.  It's as though the moves in the buffer at the time of the pause were discarded.

```
; pause.g
; called when a print from SD card is paused
M83            ; relative extruder moves
G91            ; relative positioning
G1 X5 Y50 F3600
G1 Z50 E-5 F600   
G90
```

```
; resume.g
; called before a print from SD card is resumed
M83                  ; relative extruder moves
G1 E5 F100         ; extrude 5mm of filament
```

chrishamm, 2020-04-14:
Thanks, I could reproduce this problem and it should be fixed in the latest commit.

gtjoseph, 2020-04-14:
Sort of fixed.  When resuming, the tool moves to the proper place but then extrudes a straight line to some "future" point, then draws it back again and continues with the print.  So it's not skipping the moves but it is now doing 2 extra moves.

gtjoseph, 2020-04-16:
Still an issue with commit 73334ec91505bc8d005c32a88623b2c3fc1c7945

wilriker, 2020-04-16:
I'll try to reproduce this tomorrow. Can you share the GCode file you are printing along with the rough position when to pause/resume?

chrishamm, 2020-04-16:
When I tried to reproduce this issue the other day I could not see any unwanted/future codes being sent to RRF from DSF so I'll review the firmware once more tomorrow. Did you see any duplicate or unwanted codes in the in the logs? Either way, I hope we can sort this out tomorrow - I'd really like to get v1.3.3 out ASAP.

gtjoseph, 2020-04-16:
I'll see if I can create a simple gcode file to use as a test case.

gtjoseph, 2020-04-17:
Just tested again with latest commit 8eb0f30a0dd95e53bd32e21ff10753f097dfe389

[pause-resume-test.zip](https://github.com/chrishamm/DuetSoftwareFramework/files/4490296/pause-resume-test.zip)

The gcode file prints just the first 3 loops of a skirt for a mask.  You can see 2 places where I hit pause.  On resume, the straight lines were printed, out and back, then the print resumed.

[image]

Make sure you alter the gcode for your printer filament!

wilriker, 2020-04-17:
I can reproduce the issue. Next up bug-hunt.

chrishamm, 2020-04-17:
It's fixed in the latest RRF commit. Thanks once more for helping us track down this bug!

gtjoseph, 2020-04-17:
Confirmed fixed.  Thanks!

closed 2020-04-17 by gtjoseph