Duet3D/DuetSoftwareFramework #115: Simulated time not shown for jobs

gloomyandy, 2020-04-15:
I'm not sure if this is a bug or not, but with RC7/1.3.2/2.1.2 if I run a simulation on a job (which seems to work fine), the information displayed for the job (in the jobs listing) still contains n/a for the simulated time rather than the time obtained by running the simulation. I tried a refresh but this made no difference. Should the simulated time be displayed?

chrishamm, 2020-04-15:
This is a known issue, see https://github.com/chrishamm/DuetSoftwareFramework/tree/dev#known-issues
```
Print/Simulation times are not written to G-code files yet
```

gloomyandy, 2020-04-15:
Sorry Chris, please feel free to close this issue.

chrishamm, 2020-09-03:
Simulated times will be written to the job files in DSF 3.2.

closed 2020-09-03 by chrishamm