Duet3D/DuetSoftwareFramework #116: DCS 1.3.2+1 Macros totally broken including those invoked by RRF

gtjoseph, 2020-04-15:
```
[debug] HTTP: Sent G28 Y, remaining space 1504, needed 32
[debug] Completed G28 X => Error: Homing failed
[debug] IPC#12: Connection closed
[info] Starting macro file homey.g on channel HTTP
[debug] Waiting for execution of ; homey.g (macro code)
[debug] Waiting for execution of G91 (macro code)
[debug] Waiting for execution of G1 H2 Z2 F200 (macro code)
[debug] Waiting for execution of G1 H1 Y4 F600 (macro code)
[debug] Waiting for execution of G1 H1 Y-500 F6000 (macro code)
[debug] Waiting for execution of G1 Y4 F600 (macro code)
[debug] Waiting for execution of G1 H1 Y-10 F600 (macro code)
[debug] Waiting for execution of G1 H2 Z-2 F200 (macro code)
[debug] Processing ; homey.g
[debug] Completed ; homey.g
[debug] Processing G91
[debug] Processing G1 H2 Z2 F200
[debug] Processing G1 H1 Y4 F600
[debug] Processing G1 H1 Y-500 F6000
[debug] Processing G1 Y4 F600
[error] Failed to read code from macro file homey.g
   System.NullReferenceException: Object reference not set to an instance of an object.
   at DuetControlServer.FileExecution.Macro.ReadCode() in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetControlServer/FileExecution/Macro.cs:line 329
[debug] Processing G1 H1 Y-10 F600
[debug] Processing G1 H2 Z-2 F200
[info] Aborted macro file homey.g
[info] HTTP: Finished macro file homey.g
[debug] HTTP: ==> Finished starting code: G28 Y => Error: Homing failed
[debug] Cancelled G91
[debug] Cancelled G1 H2 Z-2 F200
[debug] Completed G28 Y => Error: Homing failed
[error] Failed to read code from macro file homey.g
   System.NullReferenceException: Object reference not set to an instance of an object.
   at DuetControlServer.FileExecution.Macro.ReadCode() in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetControlServer/FileExecution/Macro.cs:line 329
[debug] Cancelled G1 H1 Y-10 F600
[debug] Cancelled G1 H2 Z2 F200
[debug] Cancelled G1 Y4 F600
[error] Failed to read code from macro file homey.g
   System.NullReferenceException: Object reference not set to an instance of an object.
   at DuetControlServer.FileExecution.Macro.ReadCode() in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetControlServer/FileExecution/Macro.cs:line 329
[debug] Cancelled G1 H1 Y4 F600
[debug] Cancelled G1 H1 Y-500 F6000
[error] Failed to read code from macro file homey.g
   System.NullReferenceException: Object reference not set to an instance of an object.
   at DuetControlServer.FileExecution.Macro.ReadCode() in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetControlServer/FileExecution/Macro.cs:line 329
[error] Failed to read code from macro file homey.g
   System.NullReferenceException: Object reference not set to an instance of an object.
   at DuetControlServer.FileExecution.Macro.ReadCode() in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetControlServer/FileExecution/Macro.cs:line 329
[error] Failed to read code from macro file homey.g
   System.NullReferenceException: Object reference not set to an instance of an object.
   at DuetControlServer.FileExecution.Macro.ReadCode() in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetControlServer/FileExecution/Macro.cs:line 329
[error] Failed to read code from macro file homey.g
   System.NullReferenceException: Object reference not set to an instance of an object.
   at DuetControlServer.FileExecution.Macro.ReadCode() in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetControlServer/FileExecution/Macro.cs:line 329
[error] Failed to read code from macro file homey.g
   System.NullReferenceException: Object reference not set to an instance of an object.
   at DuetControlServer.FileExecution.Macro.ReadCode() in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetControlServer/FileExecution/Macro.cs:line 329
[debug] Finished codes from macro file homey.g
```

gtjoseph, 2020-04-15:
Not sure if this is related but there was a warning when building the DCS...

InputGpioPin.cs(98,26): warning CS1572: XML comment has a param tag for 'disposing', but there is no parameter by that name [/usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/LinuxDevices/LinuxDevices.csproj]

chrishamm, 2020-04-15:
Confirmed, I am working on it. I'm going to provide a new commit shortly.

chrishamm, 2020-04-15:
It's fixed in the new commit. The warning message there doesn't really matter, the error was somewhere else.

gtjoseph, 2020-04-16:
Confirmed fixed.

closed 2020-04-16 by gtjoseph