Duet3D/DuetSoftwareFramework #117: DCS 1.3.2+2 M999 Stops DCS???

gtjoseph, 2020-04-16:
I don't think that should be the default behavior and I think it needs to be in the config.json.  

If you stop it, how would a normal person restart it again?  Are you going to add a route to the webserver so it can be restarted from UIs?   If so, you should add DCS start stop and restart commands to the webserver routes.  Or maybe a generic "execute system command".

T3P3, 2020-04-16:
DCS auto restarts when run as a systemd unit. this was requested to make the whole system (RRF,DCS) act in a logical way to the user. Having it settable in DCS config so you can disable this behaviour if you are running DCS from a console would be a good idea. Lets discuss further in the forum.

chrishamm, 2020-04-16:
I've started a discussion on the forums at https://forum.duet3d.com/topic/15634/should-m999-terminate-the-dsf-core-application

Using the latest commit you can now configure the default behavior of M999 in config.json and by using the '-r' command line argument you can disable the program termination again. We had some internal discussions about this [PR](https://github.com/chrishamm/DuetSoftwareFramework/pull/106) which led to this change in the first place.

gtjoseph, 2020-04-16:
Comments back on the forum.

closed 2020-04-16 by gtjoseph