Duet3D/DuetSoftwareFramework #118: DCS Pre-1.3.3 "Z Probe already triggered" message not propagating through DCS

gtjoseph, 2020-04-16:
On a G28 Z with the probe already triggered, I get...
```
Error: Code has been cancelled
Error: G0/G1: insufficient axes homed
```
but NOT the 
```
Error: Z probe already triggered at start of probing move
```

```
[debug] IPC#164: Got new UNIX connection, checking mode...
[debug] IPC#164: Command processor added
[debug] IPC#164: Received command SimpleCode
[debug] Waiting for execution of G28 Z
[debug] Processing G28 Z
[debug] HTTP: Sent G28 Z, remaining space 1504, needed 32
[info] Starting macro file homez.g on channel HTTP
[debug] Waiting for execution of G91 (macro code)
[debug] Waiting for execution of G1 H2 Z5 (macro code)
[debug] Waiting for execution of G90 (macro code)
[debug] Processing G91
[debug] Waiting for execution of G30 (macro code)
[debug] Processing G1 H2 Z5
[debug] Processing G90
[debug] Waiting for execution of G1 Z5 (macro code)
[debug] Processing G30
[debug] Processing G1 Z5
[debug] HTTP: Sent G91, remaining space 1512, needed 24
[debug] HTTP: Sent G1 H2 Z5, remaining space 1472, needed 40
[debug] HTTP: Sent G90, remaining space 1448, needed 24
[debug] HTTP: Sent G30, remaining space 1424, needed 24
[debug] HTTP: Sent G1 Z5, remaining space 1392, needed 32
[debug] Completed G91
[debug] Completed G1 H2 Z5
[debug] Completed G90
[info] HTTP: Optional macro files deployprobe0.g and deployprobe.g not found
[debug] HTTP: Suspending code G1 Z5
[info] HTTP: Finished intermediate macro file deployprobe.g
[info] HTTP: Finished macro file deployprobe.g
[info] HTTP: Optional macro files retractprobe0.g and retractprobe.g not found
[info] HTTP: Finished macro file retractprobe.g
[debug] HTTP: ==> Finished starting code: G30
[debug] HTTP: Sent G1 Z5, remaining space 1504, needed 32
[debug] HTTP: -> Resumed suspended code
[debug] Completed G30 => Error: Z probe already triggered at start of probing move
[info] Received file abort request on channel HTTP for all files
[info] Aborted macro file homez.g
[debug] HTTP: ==> Unfinished starting code: G28 Z
[debug] Cancelled G1 Z5
[debug] Finished codes from macro file homez.g
[debug] Cancelled G28 Z
[debug] IPC#164: Connection closed
[warn] HTTP: Out-of-order reply: 'G0/G1: insufficient axes homed'
[error] G0/G1: insufficient axes homed
```

chrishamm, 2020-04-16:
Thanks for the report, it should be fixed in the latest commit but I won't be able to test it before tomorrow.

gtjoseph, 2020-04-16:
I'm getting the message now but have one other oddity...

homez.g
```
G91
G1 H2 Z5
G90
G30
G1 Z5
```
gives
```
Error: Z probe already triggered at start of probing move
G0/G1: insufficient axes homed
```
Which makes sense since the `G1 Z5` doesn't have the H2 parameter to ignore homed state.

But... If I remove the `G1 Z5` I get...
```
Error: Z probe already triggered at start of probing move
Error: Homing failed
```
Should the `Error: Homing failed` message also appear in the first example?
Maybe the better question is... Should homing macros continue after encountering an error?

gtjoseph, 2020-04-16:
One final message oddity.  Messages like the bed levelling results are only going to the UI that started the command.  I.E.  If I initiate the bed levelling from a browser on the SBC, then walk back to my desk, the browser on the desktop doesn't get the completion message.

dc42, 2020-04-17:
> One final message oddity. Messages like the bed levelling results are only going to the UI that started the command. I.E. If I initiate the bed levelling from a browser on the SBC, then walk back to my desk, the browser on the desktop doesn't get the completion message.

That's normal behaviour of RRF. Messages relating to particular commands are only sent to the input channel that sent the command; except that if the channel is a print job from the SD card then it is sent to all channels.

chrishamm, 2020-04-17:
Yes you are right, the G1 Z5 code should not be sent in the first place. I'll fix that.

chrishamm, 2020-04-17:
This issue has been fixed in the latest commit.

gtjoseph, 2020-04-17:
> That's normal behaviour of RRF. Messages relating to particular commands are only sent to the input channel that sent the command; except that if the channel is a print job from the SD card then it [...]

Was it always like that?  I don't remember.  From an RRF perspective, is the SBC one channel or is RRF aware of the channels beyond like multiple UI instances?   @chrishamm would it be possible after things settle down to get an option in the DCS to 'broadcast' command responses?

chrishamm, 2020-04-17:
Hmm, it would be possible to output code results as generic messages (which are sent everywhere) but then it's tricky to avoid duplicate messages without breaking potential plugins.

OTOH I figure the web server could be changed to pass code results of codes that it initiated to clients that have a different IP address as part of the model subscription. Not trivial either way.

gtjoseph, 2020-04-17:
No worries.  I'll think more on it and bring it up in the forums if needed.

closed 2020-04-17 by gtjoseph