Duet3D/DuetSoftwareFramework #124: DCS Pre 2.1.1:  Last commit spits out lots of JSON in command responses

gtjoseph, 2020-04-22:
Commit b715f060d8a0a77e3f50f5c40616f4036c7401d9

Many commands are getting responses like: 
```
{"key":"","flags":"vd20","result":{"boards":[{"canAddress":0,"firmwareDate":"2020-04-22b1","firmwareFileName":"Duet3Firmware_MB6HC.bin","firmwareName":"RepRapFirmware for Duet 3 MB6HC","firmwareVersion":"3.01-RC9+1","iapFileNameSBC":"Duet3_SBCiap_MB6HC.bin","iapFileNameSD":"Duet3_SDiap_MB6HC.bin","maxHeaters":32,"maxMotors":6,"mcuTemp":{"current":41.1,"max":41.6,"min":27.2},"name":"Duet 3 MB6HC","shortName":"MB6HC","supports12864":false,"v12":{"current":12.2,"max":12.3,"min":0.3},"vIn":{"current":25.4,"max":25.8,"min":0.1}}],"directories":{"filaments":"0:/filaments/","firmware":"0:/sys/","gCodes":"0:/gcodes/","macros":"0:/macros/","menu":"0:/menu/","scans":"0:/scans/","system":"0:/sys/","web":"0:/www/"},"fans":[{"actualValue":0,"blip":0.10,"max":1.00,"min":0.10,"name":"PartCooling","requestedValue":0,"rpm":-1,"thermostatic":{"heaters":[]}},{"actualValue":0,"blip":0.10,"max":1.00,"min":0.10,"name":"HotEnd","requestedValue":1.00,"rpm":-1,"thermostatic":{"heaters":[1],"highTemperature":45.0,"lowTemperature":45.0}},{"actualValue":0.38,"blip":0.10,"max":1.00,"min":0.10,"name":"PSU","requestedValue":0.38,"rpm":-1,"thermostatic":{"heaters":[]}},null,null,null,null,null,null,null,{"actualValue":1.00,"blip":0,"max":1.00,"min":0.10,"name":"Bed LED","requestedValue":1.00,"rpm":-1,"thermostatic":{"heaters":[]}},{"actualValue":1.00,"blip":0,"max":1.00,"min":0.10,"name":"Front LED","requestedValue":1.00,"rpm":-1,"thermostatic":{"heaters":[]}}],"heat":{"bedHeaters":[0,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"chamberHeaters":[-1,-1,-1,-1],"coldExtrudeTemperature":160.0,"coldRetractTemperature":90.0,"heaters":[{"active":70.0,"current":26.8,"max":125.0,"min":-273.1,"model":{"deadTime":2.8,"enabled":true,"gain":96.1,"inverted":false,"maxPwm":1.00,"pid":{"d":0.5,"i":0.0,"overridden":false,"p":0.2,"used":true},"standardVoltage":0,"timeConstant":89.3},"monitors":[{"action":0,"condition":"tooHigh","limit":125.0},{"condition":"disabled"},{"condition":"disabled"}],"sensor":0,"standby":55.0,"state":"off"},{"active":225.0,"current":26.8,"max":280.0,"min":-273.1,"model":{"deadTime":5.1,"enabled":true,"gain":805.4,"inverted":false,"maxPwm":1.00,"pid":{"d":0.2,"i":0.0,"overridden":false,"p":0.1,"used":true},"standardVoltage":25.4,"timeConstant":312.7},"monitors":[{"action":0,"condition":"tooHigh","limit":280.0},{"condition":"disabled"},{"condition":"disabled"}],"sensor":1,"standby":160.0,"state":"off"}]},"inputs":[{"axesRelative":false,"compatibility":"RepRapFirmware","distanceUnit":"mm","drivesRelative":false,"feedRate":50.0,"inMacro":false,"lineNumber":0,"name":"HTTP","stackDepth":0,"state":"executing","volumetric":false},{"axesRelative":false,"compatibility":"Marlin","distanceUnit":"mm","drivesRelative":false,"feedRate":50.0,"inMacro":false,"lineNumber":0,"name":"Telnet","stackDepth":0,"state":"idle","volumetric":false},{"axesRelative":false,"compatibility":"RepRapFirmware","distanceUnit":"mm","drivesRelative":false,"feedRate":50.0,"inMacro":false,"lineNumber":0,"name":"File","stackDepth":0,"state":"idle","volumetric":false},{"axesRelative":false,"compatibility":"Marlin","distanceUnit":"mm","drivesRelative":false,"feedRate":50.0,"inMacro":false,"lineNumber":0,"name":"USB","stackDepth":0,"state":"idle","volumetric":false},{"axesRelative":false,"compatibility":"RepRapFirmware","distanceUnit":"mm","drivesRelative":false,"feedRate":50.0,"inMacro":false,"lineNumber":0,"name":"Aux","stackDepth":0,"state":"idle","volumetric":false},{"axesRelative":false,"compatibility":"RepRapFirmware","distanceUnit":"mm","drivesRelative":false,"feedRate":50.0,"inMacro":false,"lineNumber":0,"name":"Trigger","stackDepth":0,"state":"idle","volumetric":false},{"axesRelative":false,"compatibility":"RepRapFirmware","distanceUnit":"mm","drivesRelative":false,"feedRate":50.0,"inMacro":false,"lineNumber":9,"name":"Queue","stackDepth":0,"state":"idle","volumetric":false},{"axesRelative":false,"compatibility":"RepRapFirmware","distanceUnit":"mm","drivesRelative":false,"feedRate":50.0,"inMacro":false,"lineNumber":0,"name":"LCD","stackDepth":0,"state":"idle","volumetric":false},{"axesRelative":false,"compatibility":"RepRapFirmware","distanceUnit":"mm","drivesRelative":false,"feedRate":50.0,"inMacro":false,"lineNumber":0,"name":"SBC","stackDepth":0,"state":"idle","volumetric":false},{"axesRelative":false,"compatibility":"RepRapFirmware","distanceUnit":"mm","drivesRelative":false,"feedRate":50.0,"inMacro":false,"lineNumber":0,"name":"Daemon","stackDepth":0,"state":"idle","volumetric":false},{"axesRelative":false,"compatibility":"RepRapFirmware","distanceUnit":"mm","drivesRelative":false,"feedRate":50.0,"inMacro":false,"lineNumber":0,"name":"Autopause","stackDepth":0,"state":"idle","volumetric":false}],"job":{"file":{"filament":[],"firstLayerHeight":0,"height":0,"layerHeight":0,"numLayers":0,"size":0},"filePosition":0,"lastDuration":0,"timesLeft":{}},"limits":{"axes":10,"axesPlusExtruders":20,"bedHeaters":12,"boards":21,"chamberHeaters":4,"drivers":26,"driversPerAxis":5,"extruders":16,"extrudersPerTool":8,"fans":16,"gpInPorts":16,"gpOutPorts":16,"heaters":32,"heatersPerTool":8,"monitorsPerHeater":3,"restorePoints":6,"sensors":56,"spindles":2,"tools":50,"trackedObects":30,"triggers":32,"volumes":1,"workplaces":9,"zProbeProgramBytes":8,"zProbes":4},"move":{"axes":[{"acceleration":3600.0,"babystep":0,"current":1750,"drivers":["0.0"],"homed":true,"jerk":1600.0,"letter":"X","machinePosition":230.000,"max":235.0,"maxProbed":false,"microstepping":{"interpolated":true,"value":64},"min":-235.0,"minProbed":false,"speed":16000.0,"stepsPerMm":320.00,"userPosition":230.000,"visible":true,"workplaceOffsets":[0,0,0,0,0,0,0,0,0]},{"acceleration":3600.0,"babystep":0,"current":1750,"drivers":["0.1"],"homed":true,"jerk":1600.0,"letter":"Y","machinePosition":-222.000,"max":235.0,"maxProbed":false,"microstepping":{"interpolated":true,"value":64},"min":-235.0,"minProbed":false,"speed":16000.0,"stepsPerMm":320.00,"userPosition":-222.000,"visible":true,"workplaceOffsets":[0,0,0,0,0,0,0,0,0]},{"acceleration":150.0,"babystep":0,"current":1750,"drivers":["0.3","0.4","0.5"],"homed":true,"jerk":1000.0,"letter":"Z","machinePosition":4.900,"max":420.0,"maxProbed":false,"microstepping":{"interpolated":true,"value":16},"min":0,"minProbed":false,"speed":2400.0,"stepsPerMm":800.00,"userPosition":4.872,"visible":true,"workplaceOffsets":[0,0,0,0,0,0,0,0,0]}],"calibration":{"final":{"deviation":0.000,"mean":0.000},"initial":{"deviation":0.007,"mean":-0.017},"numFactors":3},"compensation":{"fadeHeight":null,"meshDeviation":{"deviation":0,"mean":0},"probeGrid":{"radius":-1.0,"xMax":235.0,"xMin":-235.0,"xSpacing":23.5,"yMax":235.0,"yMin":-235.0,"ySpacing":23.5},"type":"mesh"},"currentMove":{"acceleration":0,"deceleration":0,"requestedSpeed":0,"topSpeed":0},"daa":{"enabled":false,"minimumAcceleration":10.0,"period":0.0},"extruders":[{"acceleration":6000.0,"current":1000,"driver":"0.2","factor":1.00,"filament":"","jerk":1600.0,"microstepping":{"interpolated":true,"value":32},"nonlinear":{"a":0,"b":0,"upperLimit":0.20},"position":0,"pressureAdvance":0,"rawPosition":0,"speed":6000.0,"stepsPerMm":5400.00}],"idle":{"factor":0.5,"timeout":5.0},"kinematics":{"forwardMatrix":[[0.500,0.500,0],[0.500,-0.500,0],[0,0,1.000]],"inverseMatrix":[[0.500,0.500,0],[0.500,-0.500,0],[0,0,1.000]],"name":"coreXY"},"printingAcceleration":3600.0,"speedFactor":1.00,"travelAcceleration":3600.0,"workspaceNumber":1},"network":{"hostname":"duet3j","interfaces":[{"actualIP":"0.0.0.0","gateway":"0.0.0.0","mac":"be:76:3d:36:53:31","state":"disabled","subnet":"0.0.0.0","type":"ethernet"}],"name":"duet3j"},"sensors":{"analog":[{"lastReading":26.8,"name":"Bed","type":"thermistor"},{"lastReading":26.8,"name":"HotEnd","type":"thermistor"},{"lastReading":27.2,"name":"Bed Check","type":"thermistor"},{"lastReading":25.0,"name":"Ambient","type":"thermistor"}],"endstops":[{"triggered":false,"type":"input
```

chrishamm, 2020-04-23:
I doubt this response comes from DSF because the "vd20" flags are not used anywhere. I tried to reproduce it by sending a couple of commands but I didn't get any unexpected responses.

How can you reproduce this problem?

gtjoseph, 2020-04-23:
I think this is an RRF problem.  Responses are being concatenated into the message buffer so if I do a `M409` every command response after that will have that output.

gtjoseph, 2020-04-23:
Nope DCS.    
* Open DWC / Console and issue `M409 K"state"`.  You should get the normal state response.
* From a terminal issue `curl -s -d 'M409 F"vd20"' http://<sbc>/machine/code`.  You shold get the RRF OM.
* Back in DWC, issue `M409 K"state"` again.  You'll get the output from the curl request plus the output from the state request from DWC.

If you restart DCS, it'll clear.

gtjoseph, 2020-04-23:
Actually, you can replicate this just from DWC.  No need for curl.  I think it's related to the response length.  If it's over a certain amount, like an M409 F"vd10" would be, it gets stuck in the buffer.

wilriker, 2020-04-23:
I can confirm this behavior. Repeated `M409 K"vd10"` result in the new result being appended to the old one(s).

chrishamm, 2020-04-24:
Thanks, I'll fix it in the next DSF commit.

chrishamm, 2020-04-24:
This bug and a few other things should be fixed in the latest commit.

gtjoseph, 2020-04-24:
Confirmed fixed.

closed 2020-04-24 by gtjoseph