Duet3D/DuetSoftwareFramework #126: DCS 2.x.x: Heightmap info missing from move.compensation.

gtjoseph, 2020-04-23:
Not sure where you are with the OM but the heightmap file name and meshDeviation are missing.

```
         "compensation" : {
            "fadeHeight" : null,
            "file" : null,
            "meshDeviation" : {
               "deviation" : 0,
               "mean" : 0
            },
            "probeGrid" : {
               "radius" : -1,
               "xMax" : 235,
               "xMin" : -235,
               "xSpacing" : 23.5,
               "yMax" : 235,
               "yMin" : -235,
               "ySpacing" : 23.5
            },
            "type" : "mesh"
         },
```

chrishamm, 2020-04-24:
Thanks for the note, it seems like move.compensation.file was always overwritten from RRF OM updates and meshDeviation wasn't updated when a heightmap was loaded from the SBC. I'll fix that in the next DSF and RRF commits.

gtjoseph, 2020-04-24:
Confirmed fixed in latest RRF commit.

closed 2020-04-24 by gtjoseph