Duet3D/DuetSoftwareFramework #127: DSF and recent versions of Prusa Slicer

gloomyandy, 2020-04-23:
Recent versions of Prusa Slicer (I'm using version 2.2.0) insert the estimated print time in the form of a comment that looks like this:

```
; estimated printing time (normal mode) = 38m 31s
```

dsf 2.1.0 contains the following regular expression:
```
"estimated printing time = ((?<h>(\d+))h\s*)?((?<m>(\d+))m\s*)?((?<s>(\d+))s)?"
```
but this does not work with recent versions of PS (because of the extra "(normal mode)", not inserted in earlier versions). A modified version that does match (and that should still match older strings) is:
```
"estimated printing time .*= ((?<h>(\d+))h\s*)?((?<m>(\d+))m\s*)?((?<s>(\d+))s)?"
```
Note the additional ".*" just before "="

In addition it is common for the start up gcode used with this version of PS to contain the following line:
```
M221 S{if layer_height<0.075}100{else}95{endif}
```
This line gets included in the gcode file (as a comment) by PS, but it causes a false match against the DSF layer height matching regular expression:
```
"\slayer_height\D+(?<mm>(\d+\.?\d*))"
```
and results in the layer height being shown as 0.07mm rather than the correct 0.20mm. The following modified version seems to avoid the false positive and returns the correct layer height:
```
";\slayer_height\D+(?<mm>(\d+\.?\d*))"
```
Note the extra ";" at the start of the string.

I've tested the above modified expressions with a number of versions of PS (going back a year or so) and it seems to match values correctly in generated files.

chrishamm, 2020-04-24:
Thanks, I'll add your first updated RegEx to DSF. However I won't add the second and instead only apply regexes to found comments, that should resolve the latter problem too.

gloomyandy, 2020-04-24:
HI, I'm not sure exactly what you mean by only apply them to "found comments", but if you only intend to apply regex to comment lines then this will still trigger a false positive as the line "M221 S{if layer_height<0.075}100{else}95{endif}" is included in a comment that lists all of the startup gcode. It is that comment that triggers the current false positive (even though the line contains lots of stuff). So for instance this is the full line that causes a match with the current regex...
```
; start_gcode = G90 ; use absolute coordinates\nM83 ; extruder relative mode\nM104 S[first_layer_temperature] ; set extruder temp\nM140 S[first_layer_bed_temperature] ; set bed temp\nM190 S[first_layer_bed_temperature] ; wait for bed temp\nM109 S[first_layer_temperature] ; wait for extruder temp\nG28 W ; home all without mesh bed level\nG29 S1 ; mesh bed leveling\nM572 D0 S0.09\nG1 Y-3.0 F1000.0 ; go outside print area\nG92 E0.0\nG1 X60.0 E9.0 F1000.0 ; intro line\nG1 X100.0 E12.5 F1000.0 ; intro line\nG92 E0.0\nM221 S{if layer_height<0.075}100{else}95{endif}
```
As you can see the M221 line is embedded right at the end!

chrishamm, 2020-04-24:
Yes that's true so I will change the regex to

```
^\s*layer_height\D+(?<mm>(\d+\.?\d*))
```

Note that the the `;` will be stripped from the internal comment representation.

chrishamm, 2020-04-27:
This should be fixed in DSF 2.1.1.

gloomyandy, 2020-05-01:
Yep seems fine to me. Thanks!

closed 2020-05-01 by gloomyandy