Duet3D/DuetSoftwareFramework #129: echo breaks on comma (,)

mvieleers, 2020-04-29:
Hi, first of all, so very happy that conditional gcodes made it to dsf. Thank you for that.

While running my conditional gcode for bed levelling (that works on SD setup) on SBC it throws an error when executing the echo command.

This:
echo "Auto calibration successful, deviation: " ^ move.calibration.final.deviation ^ "mm"
Throws error:
Error: Failed to evaluate ""Auto calibration successful. deviation": Unterminated quotes

This:
echo "Auto calibration successful. deviation: " ^ move.calibration.final.deviation ^ "mm"
Works

It trips on the comma after "successful".

chrishamm, 2020-04-29:
Thanks for reporting this, it will be fixed in the upcoming DSF 2.1.3.

chrishamm, 2020-04-30:
Fixed in DSF 2.1.3.