Duet3D/DuetSoftwareFramework #130: DCS 2.1.3: M500 writes invalid G10 command to config-override.g

gtjoseph, 2020-05-01:
```
; config-override.g file generated in response to M500 at 2020-05-01 08:44

; Heater model parameters
M307 H0 A96.1 C89.3 D2.8 S1.0 V0.0 B0
M307 H1 A812.8 C306.6 D4.1 S1.0 V25.4 B0
; Workplace coordinates
G10 L2 P0 X0.00 Y0.00 Z0.00
G10 L2 P1 X0.00 Y0.00 Z0.00
G10 L2 P2 X0.00 Y0.00 Z0.00
G10 L2 P3 X0.00 Y0.00 Z0.00
G10 L2 P4 X0.00 Y0.00 Z0.00
G10 L2 P5 X0.00 Y0.00 Z0.00
G10 L2 P6 X0.00 Y0.00 Z0.00
G10 L2 P7 X0.00 Y0.00 Z0.00
G10 L2 P8 X0.00 Y0.00 Z0.00
```
The `G10 L2 P0 X0.00 Y0.00 Z0.00` line reports
Error: G10: Bad or missing parameter
The other lines complete without error.

Here's my tool definition...
```
; Tools
M563 S"HotEnd" P0 D0 H1
G10 P0 R160 S225
```

chrishamm, 2020-05-01:
Thanks, I'll check this next week.

chrishamm, 2020-05-04:
I can confirm this bug and I will fix it in the next version. G10 L2 Pnnn is offset by 1 when written by RRF.

chrishamm, 2020-05-06:
Should be fixed in v2.2.0.

gtjoseph, 2020-05-06:
Confirmed!

closed 2020-05-06 by gtjoseph