Duet3D/DuetSoftwareFramework #131: M117 Not parsing string replacements

gtjoseph, 2020-05-02:
When:
 `M117 { "Axis " ^ ( move.axes[0].letter ) ^ " not homed. Please wait while all axes are homed" }`
is executed from a Duet3 serial console the following message is received:
`Axis X not homed. Please wait while all axes are homed`

When the same command is executed via DCS, the following message is received:
`{ "Axis " ^ ( move.axes[0].letter ) ^ " not homed. Please wait while all axes are homed" }`

M291 does seem to work.
`M291 P{ "Axis " ^ (move.axes[0].letter) ^ " not homed. Continue?"} R"Check axis" S1`
 executed via DCS does cause a popup with `Axis X not homed. Continue?`

chrishamm, 2020-05-04:
This is a bug in RRF that will be adressed in the next version.

chrishamm, 2020-05-06:
Definitely fixed in v2.2.0.

closed 2020-05-06 by chrishamm