Duet3D/DuetSoftwareFramework #132: DCS 2.1.3: the "iterations" meta command is not handled correctly

bluesign2k, 2020-05-03:
The "iterations" meta command variable isn't handled correctly when using a D3+SBC setup. This was discovered when attempting to use the example object identification macro posted by dc42 [here](https://forum.duet3d.com/topic/15648/macro-to-assist-with-using-object-cancellation/1).
The provided macro example works as intended when the D3 is used in standalone mode, but returns an error in a D3+SBC setup - see [this post](https://forum.duet3d.com/topic/15648/macro-to-assist-with-using-object-cancellation/11) for further details.

chrishamm, 2020-05-04:
Thanks, I'll fix it in the next version. You can work-around this problem by putting expressions in square brackets into curly brackets like

```
move.axes[{iterations}].homed
```

chrishamm, 2020-05-06:
Fixed in v2.2.0.

closed 2020-05-06 by chrishamm