Duet3D/DuetSoftwareFramework #133: DCS 2.1.3: problem with interpreting labels for identifying objects

bluesign2k, 2020-05-03:
Comment labels for identifying objects in both PrusaSlicer and Simplify3D (as processes) don't seem to be being passed to RRF correctly.
Currently if a object-labelled gcode file from PS or a multi-process gcode file from S3D is run on a D3 in standalone mode, the objects are detected correctly. However, if the same files are run on a D3+SBC setup through DSF, no objects are identified.
See [here](https://forum.duet3d.com/topic/15648/macro-to-assist-with-using-object-cancellation/6) for further information.

chrishamm, 2020-05-04:
This is correct and a known issue (see [WHATS_NEW](https://github.com/chrishamm/DuetSoftwareFramework/blob/dev/WHATS_NEW.md#version-213) or DSF release notes). It will be implemented in the next version.

bluesign2k, 2020-05-04:
Yup, sorry, my bad - I'd only read the release notes for RRF, not DSF.
Looking forward to the next update!

chrishamm, 2020-05-06:
Should be fixed in v2.2.0.

closed 2020-05-15 by chrishamm