Duet3D/DuetSoftwareFramework #134: Allow SPI mode and buffer size to be set by config

gloomyandy, 2020-05-07:
Hi,
first off this is not really an issue, more a question/request. 

At the moment the LPC version of RRF requires two changes to the SBC source to allow DSF to work with an LPC based board. The first change is the SPI operating mode (from 0 to 3) as unfortunately the LPC SPI module is not compatible with the rPi SPI module when in mode 0 (or mode 2). The second change is to the SPI transfer buffer size (from 8192 bytes to 4096 as we simply don't have enough RAM on the LPC). I wonder if it might be possible to make these two settings part of the external configuration (there are already a number of buffer size etc settings in there). Doing that would mean we no longer need to build a modified version.

It's not a huge issue (as there are not that many LPC/SBC users), but it would make things easier.

jaysuk, 2020-05-14:
+1 for this please!

gtjoseph, 2020-05-26:
I've submitted a change to @chrishamm that adds 2 new settings to the DCS...  "SpiTransferMode" which can have a value of 0-3, and "SpiBufferSize" which can have any value but shouldn't have a value greater than the kernel's spidev buffer size.  The defaults are as today: 0 and 8192.

If you want to test, you need to already be on DSF 3.1.1 then unzip the attached file in "/opt/dsf/bin" and restart both duetcontrolserver and duetwebserver.  Make sure things still work as expected _before_ updating your /opt/dsf/conf/config.json file, then try the changes.

[dcs-3.1.1-spi.zip](https://github.com/Duet3D/DuetSoftwareFramework/files/4682601/dcs-3.1.1-spi.zip)

chrishamm, 2020-07-15:
@gtjoseph's PR has been merged, so you can expect it to be included in v3.2.

PR #146 (2020-07-15): DCS: Add more control of SPI transfers
* Added DCS parameter SpiBufferSize to allow altering the buffer
  size for small memory footprint devices.

* Added DCS parameter SpiTransferMode to allow setting the
  transfer mode for alternate platforms.

Fixes #134
closed 2020-07-15 by chrishamm