Duet3D/DuetSoftwareFramework #135: DCS 2.2.0: Max response from gcode commands seems to be 10240 bytes

gtjoseph, 2020-05-10:
If you issue a `M409 F"v,n,d5"` the response is cut off at 10240 bytes.  Issuing the same command to the Duet via USB returns a full response.

chrishamm, 2020-05-11:
This is correct. RepRapFirmware on the SAME70 (Duet 3) is configured to have 40 output buffers with 256 bytes each = 10.240 bytes. 

If you want to query the full object model, I strongly recommend using http://duet3/machine/status instead of M409 because this limitation isn't present on the Pi. In fact both DWC (when connected to RRF) and DCS poll only parts of the object model to stay within these limits.

I guess RRF handles OM outputs differently when connected over USB so it's fully written. If this is a big deal for you, I suggest you open a thread on the forum and ask for better error handling when long responses from RRF are truncated.

closed 2020-05-11 by chrishamm

gtjoseph, 2020-05-11:
Nah, it was more of an observation I guess.  I use M409 only to quickly check differences between the RRF and DCS object models and just happened to try a full dump yesterday.  In standalone mode, I use rr_model for real info.