Duet3D/DuetSoftwareFramework #136: DCS 2.2.0: Messages in quick succession aren't always received by model observers including DWC

gtjoseph, 2020-05-12:
If you run the following, the only message you _may_ see is the "heightmap loaded" message and if you add some delays in various places, you may see the "Message xl" messages and NOT the "heightmap loaded" message.

```
M117 "Printer Prepare Started"
G29 S2
M117 "Message 1"
M117 "Message 2"
M117 "Message 3"
G29 S1
M117 "Printer Prepare Completed."
```

```
 IPC#73: Command processor added
 IPC#73: Received command SimpleCode
 Waiting for execution of M98 P"/macros/PrepPrinter2.g"
 Processing M98 P"/macros/PrepPrinter2.g"
 Waiting for finish of M98 P"/macros/PrepPrinter2.g"
 HTTP: Sent M98 P"/macros/PrepPrinter2.g", remaining space 1992, needed 56
Starting macro file /macros/PrepPrinter2.g on channel HTTP
 Waiting for execution of M117 "Printer Prepare Started" (macro code)
 Waiting for execution of G29 S2 (macro code)
 Processing M117 "Printer Prepare Started"
 Waiting for execution of M117 "Message 1" (macro code)
 Waiting for finish of M117 "Printer Prepare Started"
 Waiting for execution of M117 "Message 2" (macro code)
 Processing G29 S2
 Waiting for execution of M117 "Message 3" (macro code)
 Waiting for finish of G29 S2
 Waiting for execution of G29 S1 (macro code)
 Processing M117 "Message 1"
 Waiting for execution of M117 "Printer Prepare Completed." (macro code)
 Waiting for finish of M117 "Message 1"
 Processing M117 "Message 2"
 Waiting for finish of M117 "Message 2"
 Processing M117 "Message 3"
 Waiting for finish of M117 "Message 3"
 Processing G29 S1
 HTTP: Sent M117 "Printer Prepare Started", remaining space 1996, needed 52
 HTTP: Sent G29 S2, remaining space 1964, needed 32
 HTTP: Sent M117 "Message 1", remaining space 1920, needed 44
 HTTP: Sent M117 "Message 2", remaining space 1876, needed 44
 HTTP: Sent M117 "Message 3", remaining space 1832, needed 44
 Completed M117 "Printer Prepare Started"
 Completed G29 S2
 Completed M117 "Message 1"
 Completed M117 "Message 2"
 Completed M117 "Message 3"
 Requesting update of key move, seq 86 -> 87
 Requesting update of key state, seq 65 -> 69
 Waiting for finish of G29 S1 => Height map loaded from file heightmap.csv
 Completed G29 S1 => Height map loaded from file heightmap.csv
 Processing M117 "Printer Prepare Completed."
 Waiting for finish of M117 "Printer Prepare Completed."
 HTTP: Sent M117 "Printer Prepare Completed.", remaining space 1988, needed 60
 Completed M117 "Printer Prepare Completed."
 Finished codes from macro file /macros/PrepPrinter2.g
Finished macro file /macros/PrepPrinter2.g
 Updated key move
 HTTP: Finished macro file /macros/PrepPrinter2.g
 HTTP: ==> Finished starting code: M98 P"/macros/PrepPrinter2.g" => Height map loaded from file heightmap.csv
 Completed M98 P"/macros/PrepPrinter2.g" => Height map loaded from file heightmap.csv
 Updated key state
 IPC#73: Connection closed
 Requesting update of key move, seq 87 -> 89
 Requesting update of key state, seq 69 -> 70
 Updated key move
 Updated key state
```

chrishamm, 2020-05-13:
M117 messages are intended for sticky messages which are provided by a single property in the object model. Since there is always a short delay between object model updates, successive M117 calls may have no effect and the same problem exists in stand-alone operation, too. If you wish to use M117 multiple times in the same file, either use G4+M117 or M118 S.

I am closing this issue again because it isn't a direct problem in DSF and it would need a different solution in RepRapFirmware. If you cannot use one of the suggested work-arounds, feel free to open a new thread on the forum and link to this one.

closed 2020-05-13 by chrishamm

gtjoseph, 2020-05-13:
Fair enough.  M118 messages don't display in the CodeConsole BTW.  Will open a separate issue.

chrishamm, 2020-05-13:
Feel free but it is the same issue where you asked about sending code outputs to all the web clients. Dsf code handling is different to RRF because it couples outputs to codes and not to output destinations.