Duet3D/DuetSoftwareFramework #137: DCS 2.2.0: Can't keep up at 150mm/sec print speeds even with 16MHz SPI speed

gtjoseph, 2020-05-12:
In standalone mode, the Duet3/RRF 3.01-RC12 can print at 150mm/sec without issues.  With the exact same config.g and print file, printing via DCS results in stuttering when printing lots of short segments like an arc.  Increasing the SPI clock frequency to 16MHz and doubling the BufferedPrintCodes and BufferedMacroCodes doesn't help.  During the print, the SBC is running 92% _idle_.

```
top - 17:53:51 up 3 days,  1:51,  4 users,  load average: 0.34, 0.85, 1.34
Tasks: 263 total,   1 running, 262 sleeping,   0 stopped,   0 zombie
%Cpu(s):  3.9 us,  2.5 sy,  0.0 ni, 92.2 id,  0.0 wa,  1.1 hi,  0.3 si,  0.0 st
KiB Mem :  4053568 total,  1568328 free,   772392 used,  1712848 buff/cache
KiB Swap:  4882428 total,  4882428 free,        0 used.  3135564 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND                                                                 
 2781 root      20   0 3468892  92056  38116 S  19.1  2.3   1:26.41 DuetControlServ                                                         
 6017 root      20   0 20.908g 143916  51016 S   2.6  3.6  86:13.57 DuetWebServer               

```

wilriker, 2020-05-13:
I can push SPI frequency up to around 27MHz. I doubt it will help but you could try going up to that speed and try again.

chrishamm, 2020-05-13:
That's odd, when I did my benchmark tests I could print circles up to 200mm/s. Can you share the file you're trying to print? You can try to lower SpiPollDelay in config.json to increase the number of transmission per second.

gtjoseph, 2020-05-13:
@wilriker When I went from 8 to 12 there was some improvement but when I went from 12 to 16 there was no improvement.  I can try higher though.  

@chrishamm I'll try lowering the poll delay later this morning.  In the mean time, here's the gcode file.  

Oh I was checking M122 while the file was printing and there were no issues on the Linux interface but RRF was showing underruns.

```
== Diagnostics ===
RepRapFirmware for Duet 3 MB6HC version 3.01-RC12 running on Duet 3 MB6HC v0.6 or 1.0
Board ID: 08DGM-9T66A-G63SJ-6J9F4-3SD6S-1U03B
Used output buffers: 1 of 40 (23 max)
=== RTOS ===
Static ram: 154588
Dynamic ram: 163312 of which 60 recycled
Exception stack ram used: 516
Never used ram: 74740
Tasks: NETWORK(ready,2076) HEAT(blocked,1184) CanReceiv(suspended,3824) CanSender(suspended,1436) CanClock(blocked,1432) TMC(blocked,80) MAIN(running,4516) IDLE(ready,80)
Owned mutexes:
=== Platform ===
Last reset 00:17:18 ago, cause: reset button
Last software reset at 2020-05-12 16:38, reason: User, spinning module GCodes, available RAM 75372 bytes (slot 0)
Software reset code 0x0003 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x04432000 BFAR 0x00000000 SP 0xffffffff Task MAIN
Error status: 0
MCU temperature: min 41.5, current 42.0, max 42.2
Supply voltage: min 25.3, current 25.4, max 25.4, under voltage events: 0, over voltage events: 0, power good: yes
12V rail voltage: min 12.2, current 12.2, max 12.3, under voltage events: 1
Driver 0: ok, reads 37958, writes 4 timeouts 0, SG min/max 0/830
Driver 1: ok, reads 37959, writes 4 timeouts 0, SG min/max 0/718
Driver 2: standstill, reads 37958, writes 4 timeouts 0, SG min/max 0/1023
Driver 3: standstill, reads 37958, writes 4 timeouts 0, SG min/max 0/199
Driver 4: standstill, reads 37958, writes 4 timeouts 0, SG min/max 0/208
Driver 5: standstill, reads 37959, writes 4 timeouts 0, SG min/max 0/207
Date/time: 2020-05-12 17:42:28
Slowest loop: 8.89ms; fastest: 0.21ms
=== Storage ===
Free file entries: 10
SD card 0 not detected, interface speed: 37.5MBytes/sec
SD card longest read time 0.0ms, write time 0.0ms, max retries 0
=== Move ===
Hiccups: 0(0), FreeDm: 343, MinFreeDm: 318, MaxWait: 23496ms
Bed compensation in use: none, comp offset 0.000
=== MainDDARing ===
Scheduled moves: 42824, completed moves: 42769, StepErrors: 0, LaErrors: 0, Underruns: 21, 0  CDDA state: 3
=== AuxDDARing ===
Scheduled moves: 0, completed moves: 0, StepErrors: 0, LaErrors: 0, Underruns: 0, 0  CDDA state: -1
=== Heat ===
Bed heaters = 0 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1, chamberHeaters = -1 -1 -1 -1
Heater 1 is on, I-accum = 0.4
=== GCodes ===
Segments left: 0
Movement lock held by null
HTTP* is ready with "M122" in state(s) 0
Telnet is idle in state(s) 0
File* is ready with "G1 X79.810997 Y-136.828003 E0.240600" in state(s) 0
USB is idle in state(s) 0
Aux is idle in state(s) 0
Trigger* is idle in state(s) 0
Queue* is idle in state(s) 0
LCD is idle in state(s) 0
SBC is idle in state(s) 0
Daemon* is idle in state(s) 0
Aux2 is idle in state(s) 0
Autopause is idle in state(s) 0
Code queue is empty.
=== Network ===
Slowest loop: 1.00ms; fastest: 0.01ms
Responder states: HTTP(0) HTTP(0) HTTP(0) HTTP(0) HTTP(0) HTTP(0) FTP(0) Telnet(0) Telnet(0)
HTTP sessions: 0 of 8
- Ethernet -
State: disabled
Error counts: 0 0 0 0 0
Socket states: 0 0 0 0 0 0 0 0
=== CAN ===
Messages sent 2378, longest wait 0ms for type 0
=== Linux interface ===
State: 0, failed transfers: 0
Last transfer: 25ms ago
RX/TX seq numbers: 21219/37035
SPI underruns 0, overruns 0
Number of disconnects: 0
Buffer RX/TX: 328/1448-0
=== Duet Control Server ===
Duet Control Server v2.2.0.0
File:
Buffered code: G1 X78.709 Y-137.079 E0.2857
Buffered code: G1 X77.847 Y-137.239 E0.3207
Buffered code: G1 X85.481 Y-137.239 E0.6254
Buffered code: G1 X85.453 Y-134.739 E0.7252
Buffered code: G1 X87.907 Y-132.790 F9000
Buffered code: G92 E0.0000
Buffered code: G1 X89.036 Y-132.765 E0.0450 F9000
Buffered code: G1 X89.425 Y-132.494 E0.0640
Buffered code: G1 X90.265 Y-131.861 E0.1059
Buffered code: G1 X91.079 Y-131.198 E0.1478
Buffered code: G1 X91.869 Y-130.503 E0.1898
Buffered code: G1 X92.632 Y-129.780 E0.2318
Buffered code: G1 X93.367 Y-129.028 E0.2737
Buffered code: G1 X94.073 Y-128.249 E0.3157
Buffered code: G1 X94.748 Y-127.445 E0.3576
Buffered code: G1 X95.394 Y-126.614 E0.3996
Buffered code: G1 X96.007 Y-125.761 E0.4415
Buffered code: G1 X96.588 Y-124.885 E0.4835
Buffered code: G1 X97.135 Y-123.987 E0.5255
Buffered code: G1 X97.648 Y-123.070 E0.5674
Buffered code: G1 X98.126 Y-122.133 E0.6094
Buffered code: G1 X98.568 Y-121.180 E0.6513
Buffered code: G1 X98.974 Y-120.210 E0.6933
Buffered code: G1 X99.343 Y-119.226 E0.7352
Buffered code: G1 X99.675 Y-118.228 E0.7772
Buffered code: G1 X99.968 Y-117.219 E0.8192
Buffered code: G1 X100.224 Y-116.200 E0.8611
Buffered code: G1 X100.440 Y-115.171 E0.9030
Buffered code: G1 X100.618 Y-114.134 E0.9450
Buffered code: G1 X100.757 Y-113.093 E0.9870
Buffered code: G1 X100.856 Y-112.047 E1.0289
==> 1480 bytes
Code buffer space: 2264
Configured SPI speed: 16000000 Hz
Full transfers per second: 35.46
File /opt/dsf/sd/gcodes//EarLoop.gcode is selected, processing
```
[speedtest.zip](https://github.com/chrishamm/DuetSoftwareFramework/files/4622116/speedtest.zip)

gtjoseph, 2020-05-13:
Neither increasing the frequency nor decreasing the poll delay helped.

chrishamm, 2020-05-15:
My best guess is that the Linux subsystem in RRF takes quite long to calculate the CRC16 checksums. If that is the case, this problem will be resolved in RRF 3.02 when that part has got its own RTOS task. I'll do some more investigations in the near future.

gtjoseph, 2020-05-15:
No worries.  I'll keep testing as @dc42 makes commits towards RRF 3.02.

chrishamm, 2020-12-20:
Please check this again with 3.2-RC1. The performance should be notably higher compared to v3.1.

closed 2021-03-04 by chrishamm