Duet3D/DuetSoftwareFramework #138: DCS 2.2.1-pre: Can't pause a print, restart DCS, then resume

gtjoseph, 2020-05-13:
If you pause a print then restart DCS, the proper state is reported "Paused" but if you try to resume you get `Error: M24: Cannot print, because no file is selected!`.  You _can_ cancel and start over though.

chrishamm, 2020-05-14:
Why would you ever want to do that? If RRF loses connection to DCS for more than 10 seconds, it will automatically close all pending files and stop the print if there is one. This particular timeout will be drastically reduced in RRF 3.02.

gtjoseph, 2020-05-14:
Well, I've been testing a lot lately. :)
Pause print, change SPI, buffer sizes, etc. restart, resume print.

chrishamm, 2020-05-15:
Thanks, much appreciated but this issue is not a real concern because the DCS systemd unit has a restart timer of 10 seconds ;)

closed 2020-05-15 by chrishamm