Duet3D/DuetSoftwareFramework #140: DCS 2.2.1-pre: InvalidCastException with RRF f9292646

gtjoseph, 2020-05-13:
DSF: 793f78f8cb
RRF:  https://github.com/dc42/RepRapFirmware/commit/f9292646e6accd24a6f69928e707c82b6d622a41

```
[error] IPC#12: Failed to record job/build/objects[0 of 1]/y = DuetAPI.Machine.ModelCollection`1[System.Nullable`1[System.Single]] (Property)
   System.InvalidCastException: Unable to cast object of type 'DuetAPI.Machine.Build' to type 'System.Collections.Generic.Dictionary`2[System.String,System.Object]'.
   at DuetControlServer.IPC.Processors.Subscription.MachineModelPropertyChanged(Object[] path, PropertyChangeType changeType, Object value) in /usr/src/3d/dc42/workspace/DuetSoftwareFramework/src/DuetControlServer/IPC/Processors/Subscription.cs:line 359
```

chrishamm, 2020-05-14:
Please share the G-code file and perhaps your configs. I'll try to reproduce it then.

chrishamm, 2020-05-14:
PS: I've just pushed another commit to the dev branch, I hope that fixes this problem. If it doesn't, please share the G-code file and your configs.

gtjoseph, 2020-05-14:
Confirmed fixed.

closed 2020-05-14 by gtjoseph