Duet3D/DuetSoftwareFramework #142: DSF 3.1.0:  The "merge" commit in master got the tag so "dev" isn't tagged with 3.1.0

gtjoseph, 2020-05-15:
Unfortunately, it looks like the dev->master merge commit (which is only in master) got the 3.1.0 tag so there's no version tag in the dev branch.  It might be too late but do you think you could delete the tag from b6279bcd67738970d11ad957321dce7a5cc059a7 and add it to 1cb859aa623815bbe2e8853478ea5a86e07caa22 which is common to both branches?

chrishamm, 2020-05-15:
Umm, why does that matter? I'd like to keep stable releases on the master branch and unstable releases on dev. That's what I've been doing on the DWC repo for ages.

gtjoseph, 2020-05-16:
Well, the dev branch will now have some tags but not others so it could get confusing.  @dc42 does tag RRF _before_ merging to other branches so the tagged commits are all the branches.