Duet3D/DuetSoftwareFramework #143: Minor build issue.  XML files occasionally get regenerated on build leaving git tree dirty

gtjoseph, 2020-05-15:
When you make changes to modules like DuetControlServer that affect the documentation, a build after a git pull recreates the ./src/DuetControlServer/DuetControlServer.xml file.  Since it's checked in, a build leaves the git tree dirty.  If they're build artifacts, shouldn't they be in .gitignore?

chrishamm, 2020-05-15:
The .xml files are referenced by the documentation project, which is in charge of updating the HTML docs. And in the future, the DuetAPI .xml files may be used to generate a new documentation for the object model, so I'd rather leave them there for now.

gtjoseph, 2020-05-16:
I haven't been in the dotnet world for a very long time so I'm not clear on what generates what.  If I delete the xml file and build, it _is_ regenerated and what's confusing is why a build after a clean checkout is regenerating it locally.   Do you build and regen the file before you commit?  If not, that could be why it regens locally.

chrishamm, 2020-05-16:
I am quite certain I rebuilt the Release configuration before I published 3.1.0, because the XML file is only generated with that build config. But to be fair, I roll out my binaries with the Debug build config because I could not find any downsides performance-wise and stack traces only provide line numbers if built with the Debug config.

Either way, I'll rebuild both configs before we publish 3.1.1.

gtjoseph, 2020-05-16:
Ah so the debug builds don't touch the XML files?  I didn't know that.  If that's the case and there's no downside to building Debug, then I'll do Debug builds and not worry about it.

closed 2020-05-16 by gtjoseph