Duet3D/DuetSoftwareFramework #150: Non-negligible processing delay between issuing 'SimpleCode' and changing machine state to "busy"

Poofjunior, 2020-08-13:
Here's the setup. I have have a python program with two socket connections: one in command mode and one set to poll the machine at 2.5Hz in subscribe mode in a separate thread.

Upon sending a movement-related SimpleCode, it takes a noticeable amount of time (up to 1 second) for the machine model to change its state from *idle* to *busy*. This sets up a tricky situation since we might return immediately thinking the move was already processed and do something else (take picture, pipette liquids, other non-gcode driven tasks, etc.) before the machine has finished (or even started) moving.

Here's the result of sending a G0 movement command while polling the socket in another thread setup to subscribe for changes.
```
Moving to: (96.167, 79.750) # GCode has been sent across the socket
start state: idle | 1858.664
end state:   idle | 1858.967
start state: idle | 1859.085
end state:   idle | 1859.367
start state: idle | 1859.473
curr state:  busy | 1859.767 # Status now changes from idle to busy
curr state:  busy | 1860.167
curr state:  busy | 1860.568
curr state:  busy | 1860.967
curr state:  busy | 1861.367
curr state:  busy | 1861.767
end state:   idle | 1862.167
```

High level, it would be nice to have some sort of safeguard to be able to tell the program interacting with the API that it hasn't handled the message that was sent over yet such that the interacting program doesn't keep marching forward.

Here's a suggested fix.
1. Upon receiving any movement-related gcode on the socket, *immediately* update the machine to *busy* even before processing that move. That way, subsequent polls of the machine state will not return *idle" values when the machine is still in the transition from idle to busy. What's happening now is that we're polling the machine in the transition phase before it registers itself as *busy*, even though the next state transition would be *busy*.

In the meantime, if I want to *guarantee* that the machine is done moving before issuing non-gcode command in my python program (take picture with camera, etc), I need to keep a local representation of the machine position in my python program so that I can compare it to the machine model. This kinda defeats the purpose of simple asking the machine where it is.

wilriker, 2020-08-14:
Hi @Poofjunior 

What your are looking for is an Intercept Connection of mode `EXECUTED`. Whenever you need close control over command flow you have to use an Interceptor.

I reckon there is more documentation required to describe complex workflows better. Maybe this is better continued on the forum.

Poofjunior, 2020-08-14:
Huh; very cool! I didn't realize I could take advantage of Intercept Mode in this way.

It sounds like the control flow would involve setting up another socket connection in Intercept mode (*executed* "sub" mode) and then polling it upon sending a new command from another socket to block until the command is done executing?

Also based on:

> Apart from that, you can run any commands like in the command mode while a code is being intercepted.
Does that mean intercept mode also accepts sending commands like command mode from the intercept socket connection?

Happy to take it another step along the way and re-open on the forum. Thanks for the reply and feel free to close the issue in that case!

wilriker, 2020-08-14:
I think everything you want to do is already available, so, yes, please open a thread in the forum and close this ticket with a link to it so everybody stumbling across this knows where to look instead. :-)

Poofjunior, 2020-08-20:
Ok, just rewrote to use intercept mode in Execution mode and I get the same issue. I started a [forum post here](https://forum.duet3d.com/topic/18282/tighter-control-for-waiting-for-motion-commands-to-complete).

Thanks for taking a look.

closed 2020-08-20 by Poofjunior