Duet3D/DuetSoftwareFramework #152: M291 in macros will not update state.messageBox in OM

wilriker, 2020-10-16:
When a `M291 S3 P"message" R"title"` is contained inside a macro DCS will make the dialog appear in DWC but it will not update `state.messageBox` in RRFs ObjectModel and thus it will not open a dialog in PanelDue.

It works correctly if the `M291` command is send directly via code-input in DWC or e.g. `CodeConsole`.

chrishamm, 2020-10-24:
I just tried to reproduce this on my Duet 3 and I couldn't achieve that. TBH I would've been surprised it were the case, because DSF does not interpret M291/292 itself. It only deals with corresponding events. Everything else is stored in the same OM variables and it should not matter from where you query them.

Please try out my latest RRF/DSF dev versions and make sure this problem is still present. FYI I created a macro file with `M291 S3` in it, started it from DWC, and then queried the object model using the CodeConsole utility with recompiled target channel `Aux` while the message box was still open. This is what I got back from `M409 K"state" F"d99vn"`:

```
{"key":"state","flags":"d99vn","result":{"atxPower":null,"beep":null,"currentTool":-1,"displayMessage":"","gpOut":[],"laserPwm":null,"logFile":"","machineMode":"CNC","messageBox":{"axisControls":0,"message":"message","mode":3,"seq":3,"timeout":0,"title":"title"},"msUpTime":687,"nextTool":-1,"powerFailScript":"","previousTool":-1,"restorePoints":[{"coords":[0,0,0],"extruderPos":0,"feedRate":50.0,"ioBits":0,"laserPwm":null,"spindleSpeeds":[0,0],"toolNumber":-1},{"coords":[0,0,0],"extruderPos":0,"feedRate":50.0,"ioBits":0,"laserPwm":null,"spindleSpeeds":[0,0],"toolNumber":-1},{"coords":[0,0,0],"extruderPos":0,"feedRate":50.0,"ioBits":0,"laserPwm":null,"spindleSpeeds":[0,0],"toolNumber":-1},{"coords":[0,0,0],"extruderPos":0,"feedRate":50.0,"ioBits":0,"laserPwm":null,"spindleSpeeds":[0,0],"toolNumber":-1},{"coords":[0,0,0],"extruderPos":0,"feedRate":50.0,"ioBits":0,"laserPwm":null,"spindleSpeeds":[0,0],"toolNumber":-1},{"coords":[0,0,0],"extruderPos":0,"feedRate":50.0,"ioBits":0,"laserPwm":null,"spindleSpeeds":[0,0],"toolNumber":-1}],"status":"busy","time":"2020-10-24T14:47:51","upTime":1120}}
```

So it's definitely looking good on my setup. I doubt there is any difference between a code from the SBC and a true UART channel.

wilriker, 2020-11-03:
Works with latest DCS and/or RRF (wherever the problem was).

closed 2020-11-03 by wilriker