Duet3D/DuetSoftwareFramework #158: Errors generated when attempting to install DSF on Pi

BenKitesurfs, 2021-01-13:
I started with a clean install of of DuetPi Lite (from @gloomyandy wiki page: https://github.com/gloomyandy/RepRapFirmware/wiki/SBC-LPC-Stable).

The pi boots fine (I configured WPA_supplicant and added .ssh, so I'm operating headless now).

The SBC is a **Raspberry Pi 3 Model B Rev 1.2**

When I run:

**sudo apt update**

I get the following hits:
`
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease                                        
Hit:3 https://pkg.duet3d.com unstable InRelease
`
When I run:

**sudo apt upgrade**

I get:

`
The following packages will be upgraded:
  duetcontrolserver duetruntime duetsoftwareframework duettools duetwebcontrol duetwebserver reprapfirmware
7 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
6 not fully installed or removed.
Need to get 0 B/34.9 MB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
Reading changelogs... Done
dpkg: warning: files list file for package 'duetsd' missing; assuming package has no files currently installed
(Reading database ... 57840 files and directories currently installed.)
Preparing to unpack .../0-duetsoftwareframework_3.2.0-stable_armhf.deb ...
Unpacking duetsoftwareframework (3.2.0-stable) over (3.2.0) ...
To delete the "dsf" user run as root: userdel --force dsf
Preparing to unpack .../1-duettools_3.2.0-stable_armhf.deb ...
Unpacking duettools (3.2.0-stable) over (3.2.0) ...
Preparing to unpack .../2-duetcontrolserver_3.2.0-stable_armhf.deb ...
Unpacking duetcontrolserver (3.2.0-stable) over (3.2.0) ...
To delete "gpio" group run as root: groupdel gpio
Preparing to unpack .../3-duetruntime_3.2.0-stable_armhf.deb ...
Unpacking duetruntime (3.2.0-stable) over (3.2.0) ...
Preparing to unpack .../4-duetwebserver_3.2.0-stable_armhf.deb ...
Unpacking duetwebserver (3.2.0-stable) over (3.2.0) ...
Preparing to unpack .../5-duetwebcontrol_3.2.0-stable_all.deb ...
Unpacking duetwebcontrol (3.2.0-stable) over (3.2.0) ...
Preparing to unpack .../6-reprapfirmware_3.2.0-stable-1_all.deb ...
Unpacking reprapfirmware (3.2.0-stable-1) over (3.2.0-1) ...
Setting up duetruntime (3.2.0-stable) ...
Setting up duetcontrolserver (3.2.0-stable) ...
dpkg: error processing package duetcontrolserver (--configure):
 installed duetcontrolserver package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of duettools:
 duettools depends on duetcontrolserver (= 3.2.0-stable); however:
  Package duetcontrolserver is not configured yet.

dpkg: error processing package duettools (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of duetwebserver:
 duetwebserver depends on duetcontrolserver (>= 3.2.0-stable); however:
  Package duetcontrolserver is not configured yet.

dpkg: error processing package duetwebserver (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of duetsoftwareframework:
 duetsoftwareframework depends on duetcontrolserver (= 3.2.0-stable); however:
  Package duetcontrolserver is not configured yet.
 duetsoftwareframework depends on duettools (= 3.2.0-stable); however:
  Package duettools is not configured yet.
 duetsoftwareframework depends on duetwebserver (= 3.2.0-stable); however:
  Package duetwebserver is not configured yet.

dpkg: error processing package duetsoftwareframework (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of reprapfirmware:
 reprapfirmware depends on duetsoftwareframework (>= 3.2.0); however:
  Package duetsoftwareframework is not configured yet.
 reprapfirmware depends on duetcontrolserver (>= 1.2.0.0); however:
  Package duetcontrolserver is not configured yet.

dpkg: error processing package reprapfirmware (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of duetwebcontrol:
 duetwebcontrol depends on duetsoftwareframework (>= 3.2.0); however:
  Package duetsoftwareframework is not configured yet.

dpkg: error processing package duetwebcontrol (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 duetcontrolserver
 duettools
 duetwebserver
 duetsoftwareframework
 reprapfirmware
 duetwebcontrol
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

chrishamm, 2021-01-19:
If you are really starting with a fresh DuetPi Lite image then the default package feed is `stable` and not `unstable`. Since there are no differences between these two at this time there is no need to change it either.

Also, please use the forum for further support questions ([LPC](https://forum.duet3d.com/category/35/lpc) or [STM](https://forum.duet3d.com/category/36/stm))

closed 2021-01-19 by chrishamm

BenKitesurfs, 2021-01-24:
> If you are really starting with a fresh DuetPi Lite image then the default package feed is `stable` and not `unstable`. Since there are no differences between these two at this time there is no need [...]

I have used the forum and discord, I have tried everything that they have suggested, and nothing works.  I have started with a fresh version of DuetPiLite, I have tried installing DuetPi (full version with GUI), I have tried installing raspbian and then installing DSF.

I cannot start the DCS.  When I run:

sudo systemctl start duetcontrolserver

I get:

Job for duetcontrolserver.service failed because the service did not take the steps required by its unit configuration.
See "systemctl status duetcontrolserver.service" and "journalctl -xe" for details.

chrishamm, 2021-01-25:
I am still unable to reproduce this, I've just started with a brand-new Mini5+ and flashed DuetPi Lite from the pkg server (https://pkg.duet3d.com/DuetPi-lite.zip) to an SD card and after the first boot I can see DCS is running as expected:

```
pi@duet3:~ $ systemctl status duetcontrolserver
● duetcontrolserver.service - Duet Control Server
   Loaded: loaded (/lib/systemd/system/duetcontrolserver.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2021-01-05 15:07:52 GMT; 2 weeks 6 days ago
 Main PID: 373 (DuetControlServ)
    Tasks: 21 (limit: 4915)
   CGroup: /system.slice/duetcontrolserver.service
           └─373 /opt/dsf/bin/DuetControlServer

Jan 05 15:07:51 duet3 DuetControlServer[373]: Written by Christian Hammacher for Duet3D
Jan 05 15:07:51 duet3 DuetControlServer[373]: Licensed under the terms of the GNU Public License Version 3
Jan 05 15:07:52 duet3 DuetControlServer[373]: [info] Settings loaded
Jan 05 15:07:52 duet3 DuetControlServer[373]: [info] Environment initialized
Jan 05 15:07:52 duet3 DuetControlServer[373]: [info] Connection to Duet established
Jan 05 15:07:52 duet3 DuetControlServer[373]: [info] IPC socket created at /var/run/dsf/dcs.sock
Jan 05 15:07:52 duet3 systemd[1]: Started Duet Control Server.
Jan 05 15:07:52 duet3 DuetControlServer[373]: [info] Starting macro file config.g on channel Trigger
Jan 05 15:07:52 duet3 DuetControlServer[373]: [info] Finished macro file config.g
Jan 25 20:57:44 duet3 DuetControlServer[373]: [info] System time has been changed
pi@duet3:~ $
```

BenKitesurfs, 2021-01-25:
> I am still unable to reproduce this, I've just started with a brand-new Mini5+ and flashed DuetPi Lite from the pkg server (https://pkg.duet3d.com/DuetPi-lite.zip) to an SD card and after the first [...]

Thanks for taking the time to try to replicate the problem.

I got it working last night.  I have been talking to some of the guys on the Discord server, and it turns out I either had something wrong with my loom connecting the Pi to the board, or I had the wrong .bin file on the sd card (I changed both at the same time, so I don't know which one was causing it).

I apologize for wasting your time with this, and I do appreciate your help.