Duet3D/DuetSoftwareFramework #161: Exception upon installation

M-Pixel, 2021-02-06:
The following occurs during the DuetPi build process, during the installation of the `duetsoftwareframework` package through `apt`.

```
Fatal error. System.Runtime.InteropServices.SEHException (0x80004005): External component has thrown an exception.
   at System.Text.Json.JsonReaderState..ctor(System.Text.Json.JsonReaderOptions)
   at System.Text.Json.Utf8JsonReader..ctor(System.ReadOnlySpan`1<Byte>, System.Text.Json.JsonReaderOptions)
   at DuetControlServer.Settings.LoadFromFile(System.String)
   at DuetControlServer.Settings.Init(System.String[])
   at DuetControlServer.Program+<Main>d__5.MoveNext()
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.Start[[System.__Canon, System.Private.CoreLib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]](System.__Canon ByRef)
   at System.Runtime.CompilerServices.AsyncTaskMethodBuilder.Start[[System.__Canon, System.Private.CoreLib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]](System.__Canon ByRef)
   at DuetControlServer.Program.Main(System.String[])
   at DuetControlServer.Program.<Main>(System.String[])
```

chrishamm, 2021-02-06:
What SBC and Duet board are you using? Are you trying to obtain DSF from the stable or unstable package feed?

M-Pixel, 2021-02-06:
> What SBC and Duet board are you using?
This occurs during the DuetPi build process, not after the image has been deployed to the SBC.  I'm using the Docker-based build.

> Are you trying to obtain DSF from the stable or unstable package feed?
I haven't changed this, so it's whatever the default is in the current DuetPi master.  Looking at the logs, it appears to be attempting an installation of version 3.2.0.

M-Pixel, 2021-02-06:
I'm currently working on modifying DuetSoftwareFramework to rely on `dotnet-runtime-3.1` instead of baking the runtime into each program.

It's totally a shot in the dark, but SEHException in a framework class constructor smells to me of a missing DLL due to over-aggressive stripping.  Given that DuetSoftwareFramework is made of several programs, and meant to be installed on a dedicated OS, it seems more architecturally appropriate to use shared framework binaries, so the worst case scenario is that my shot in the dark doesn't fix this problem but does leave me with smaller programs that use less RAM and build faster.

If it doesn't end up working out, I'll see if I can extract any more details about the exception.  If it does, you'll have a potential solution, or if you'd prefer to stick with standalone builds, a good hint as to what the problem is.

chrishamm, 2021-02-07:
DSF requires at least .NET Core 3.1 but it uses C# 9 which isn't part of the .NET 3.1 SDK AFAIK. So perhaps try .NET 5 SDK to build it. You can use that to build and run programs that are aimed at .NET Core 3.1 too (like DSF 3.2). I'll upgrade it to .NET 5 as part of v3.3 anyway.

M-Pixel, 2021-02-08:
I remembered reading before that as long as new C# versions don't rely on new CLR versions, which is almost always the case, there's no reason that they wouldn't work on old framework versions.  However, after reviewing [the latest documentation on the matter](https://docs.microsoft.com/en-us/dotnet/csharp/language-reference/configure-language-version), they state pretty explicitly that language 9 is only expected to work on framework 5, and that going above a framework's default is likely to cause problems due to framework library API & behavior differences.

M-Pixel, 2021-02-08:
I plan to see this through this week, and provide an update as to whether the framework separation and/or upgrade to 5.0 solves this problem.  It may be a bit slow, as the next thing blocking me from testing it is the lack of `reprapfirmware` on my repo causing apt to not install `duetsoftwareframework` at all (getting this all working _with my own from-source builds_ is more important to me personally than getting it working _soon_).

chrishamm, 2021-03-02:
I cannot reproduce this problem so I'll close this issue again.

closed 2021-03-02 by chrishamm