Duet3D/DuetSoftwareFramework #163: make debs fails on unexpanded list

M-Pixel, 2021-02-07:
1. Check-out in Debian Buster
2. Install `dotnet-sdk-3.1`, `libxml2-utils`, `make`, and `rsync`
3. `make debs`

`sed: can't read /.../bin/packages/deb/duetcontrolserver_3.2.0_armhf/DEBIAN/{control,changelog}: No such file or directory`

Is there a Make plugin or configuration that's supposed to be expanding the curly-braced lists?

gtjoseph, 2021-02-07:
Is bash your shell?  If not, I can probably make those discrete commands.

M-Pixel, 2021-02-07:
> I can probably make those discrete commands.

I've already done so on my end, can send over a PR.

Bash is indeed my shell.  I'm working off of the `3.2.0` tag.

chrishamm, 2021-02-07:
Please feel free to send over a PR for this, I don't use the Makefiles. You may have better luck with the build script in the pkg directory.

gtjoseph, 2021-02-07:
Odd.  Anyway, I'll fix it up.  I've got another fix for the Makefile to push up anyay.

gtjoseph, 2021-02-07:
Pull request submitted.

PR #164 (2021-02-07): Makefile update
* Fixup RPM builds for 3.2+

* Remove bash-isms.
  For some platforms, bash isn't make's default shell which
  caused issues with the deb creation process so the
  {control,changelog} bash-isms have been removed.

Fixes #163
closed 2021-03-02 by chrishamm