Duet3D/DuetSoftwareFramework #165: bed.g code stopt working in version 3.2.2

mvieleers, 2021-02-23:
Duet3 with SBC

Duet Firmware: 3.2.2
DSF Framework: 3.22
DWC: 3.2.2

Not entirely sure this should be posted in DSF or in Duet GitHub. But I'm guessing DSF.

I've got the following bed.g:
`; bed.g
; called to perform automatic bed compensation via G32

M561                         ; clear any existing bed transform
; If the printer hasn't been homed, home it
if !move.axes[0].homed || !move.axes[1].homed || !move.axes[2].homed
  G28
  
G1 Z7.5

M98 P"/macros/4_Level and Probe/2_Probe_Bedlevel.g"

while true
  if iterations = 5
    abort "Too many auto calibration attempts"
  if move.calibration.initial.deviation < 0.01 
    if move.calibration.final.deviation < move.calibration.initial.deviation + 0.005
      if move.calibration.final.deviation > move.calibration.initial.deviation - 0.005
        break
  echo "Repeating calibration because initial deviation (" ^ move.calibration.initial.deviation ^ "mm) must be < 0.01"
  echo "and (" ^ move.calibration.final.deviation ^ "mm) must be within 0.005 of initialDeviation"

  M98 P"/macros/4_Level and Probe/2_Probe_Bedlevel.g"

echo "Auto calibration successful, deviation", move.calibration.final.deviation ^ "mm"
echo "Auto calibration successful, initialDeviation", move.calibration.initial.deviation ^ "mm"  
    
G28 Z

G1 X0 Y0 F99999               ; move the head to the corner (optional)

`

This code worked fine up until version 3.2.0
After upgrading to 3.2.2 the gcode gets stuck when in it's in the IF statement.
It will not execute the second time it calls the probing macro with M98. Instead it remains in the last position of the first time it runs 2_Probe_Bedlevel.g. The status remains in "busy" and it will not respond to any other commands anymore. I have to reset the board.

When if move.calibration.initial.deviation < 0.01 is true the flow is correct and it continues on to the echo and G28 Z gcode.

chrishamm, 2021-03-01:
Thanks for reporting this. I've been able to identify the cause for this and it turns out that your expression is too long. Please do not make your expressions longer than 99 characters. I'll enforce this limitation on the SBC in the next software version.

In your particular case your expression is truncated to
```
Repeating calibration because initial deviation (" ^ move.calibration.initial.deviation ^ "mm) must
```

mvieleers, 2021-03-01:
Gotcha. Knowing this I can work around it.
Many thanks!

chrishamm, 2021-03-04:
In v3.3-beta2 we are going to increase the maximum expression size to 256 bytes,

chrishamm, 2021-04-28:
Fixed in 3.3-b2.

closed 2021-04-28 by chrishamm