Duet3D/DuetSoftwareFramework #172: Marlin emulation not always returning command status.

NNMavy, 2021-06-13:
While investigating an issue with a TFT screen I discovered that the M20 and M503 commands do not return "ok" after executing. This resulted in the TFT waiting endlessly for the command to have ben executed.

So far I have identified M20 and M503 to be broken but I suspect more M commands in MCodes.cs will be broken when using a SBC and marlin emulation.

jaysuk, 2021-06-13:
This when using a BTT TFT screen as they have recently been adding RRF support

NNMavy, 2021-06-13:
This was with a BTT screen, however I can imagine other devices that rely on marlin mode will have a similar issue.

So far I have narrowed the issue down to [this ](https://github.com/Duet3D/DuetSoftwareFramework/blob/2dfaf25296909d8b0ed430ec462183b92a8726a8/src/DuetControlServer/Commands/Generic/Code.cs#L632)line of code. It seems that the IsLastCode flag is not set and as a result the marlin emulation is not working as expected.

chrishamm, 2021-06-15:
I think I fixed some related problems in v3.3 and I do get the proper `ok` responses below M20 and M503 so I'll close this issue again. To be safe I added another test case to the DSF unit tests and they confirm that the IsLastCode flag is properly set.

closed 2021-06-15 by chrishamm