Duet3D/DuetSoftwareFramework #176: [FeatureRequest]: Configurable path to systemctl or fallback

LoicGRENON, 2021-12-17:
### Is your feature request related to a problem? Please describe.

There is an issue when using this plugin out of a Pi (using another SBC running Armbian, for example) : `systemctl` is located at `/bin/systemctl` instead of `/usr/bin/systemctl`, on Armbian for example.

This may also apply for others files this plugin may use, I didn't checked too much deeply yet.

### Describe the solution you propose.

One solution could be to use symlinks on the SBC side but it implies the user to understand what is going on and then to mess with his SBC.

Another solution could be to set the paths used in the plugin by checking if files like `/etc/armbian-release` exists and set the paths accordingly. That would requires no action from the user.

Obviously, I'm willing to check the plugin-features on Armbian.

### Describe alternatives you've considered

Ex. A workaround exists but it is tedius, for example...

### Provide any additional context or information.

Ex. Photos, mockups, etc.

jaysuk, 2021-12-17:
just to add, this is in regards to the duetpimanagementplugin

chrishamm, 2022-01-29:
Can you explain why this is a problem? The current DSF pre/post install/removal scripts don't use absolute systemctl paths but rely on a proper $PATH setup. In all Linux distributions I know both /bin and /usr/bin are part of $PATH by default.

LoicGRENON, 2022-01-29:
DuetPiManagementPlugin do use an absolute path actually : https://github.com/Duet3D/DuetSoftwareFramework/blob/bd1f8d43fdd25cf520bbf0f4186c02e7bdc2b3a0/src/DuetPiManagementPlugin/Network/Protocols/HTTP.cs#L62

chrishamm, 2022-02-08:
Thanks for the note, I'll use relative paths in the next version.

AFAIR .NET required absolute filenames in older versions but it seems to work now with .NET 6.

chrishamm, 2022-02-10:
Now implemented in 3.4-rc1.

closed 2022-02-10 by chrishamm