Duet3D/DuetSoftwareFramework #192: Add support for decimal data type

chrishamm, 2023-10-06:
DSF API only exposes float (single precision) data at present. This needs to be enhanced to support .NET `decimal` values by default, which will also allow display of an arbitrary number of decimal digits. Support for this needs to be implemented in RRF, too.

dc42, 2023-10-06:
Why?

chrishamm, 2023-10-06:
To fix https://github.com/Duet3D/DuetSoftwareFramework/issues/187. Also, when you parse e.g. `12.000` and convert it back to a string, `12.000` is output instead of `12`. If you want to output a specific number of decimal points, that's the best way to go.

dc42, 2023-10-12:
When RRF stores a float in an ExpressionValue, the value already includes a number of significant decimal places. I think all we need to do is to have that number returned to DSF along with the  float value, whenever EvaluateExpression returns a value to DSF.