Duet3D/DuetSoftwareFramework #203: [Bug]: assigning global variables to an array using meta-commands make DSF and RepRap to crash

fragrama17, 2024-09-08:
### Duet Forum Discussion Thread

https://forum.duet3d.com/topic/36519/bug-assigning-global-variables-to-an-array-crashes-reprap

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [X] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

3.5.1

### Duet Web Control Version

None

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [X] Yes I use a SBC.
- [ ] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
M122 Report
```

### Please upload the content of your config.g file.

```gcode
Config.g
```

### Please upload the content of any other releveant macro files.

```gcode
global obj = move.axes
```

### What happened?

Observed result
dsf and reprap break producing these messages
`Sep 6, 2024, 8:52:37 AM
 Internal Error in GetIndex at ../src/ObjectModel/ObjectModel.cpp(577) 
Sep 6, 2024, 8:52:37 AM
 Internal Error in GetIndex at ../src/ObjectModel/ObjectModel.cpp(577) 
Sep 6, 2024, 8:52:37 AM
 Internal Error in GetIndex at ../src/ObjectModel/ObjectModel.cpp(577) 
Sep 6, 2024, 8:52:37 AM
 Internal Error in GetLastIndex at ../src/ObjectModel/ObjectModel.cpp(588) `

Steps to reproduce
run command: `global obj = move.axes`

chrishamm, 2024-09-09:
What's the use-case of an assignment like this? Is the bug still present in 3.5.2 stable?

fragrama17, 2024-09-10:
I've observed the same behavior in version 3.5.2.

It's important to note that the issue isn't necessarily with the feature itself, but rather that it could potentially crash the system.

Would it be possible to implement a check like it has already been done for assignment to an object ?
current behavior for an object assignment:
command -> global obj = move.axes[0]
response -> Error: Cannot assign a value of type 'object' to a variable 

Thanks for your consideration.