Duet3D/DuetSoftwareFramework #215: [Bug]: Unable to display `move.axes[].drivers` array

AndyEveritt, 2025-03-26:
### Duet Forum Discussion Thread

NA

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [x] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [x] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

3.6.0-rc2

### Duet Web Control Version

3.6.0-rc2

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [x] Yes I use a SBC.
- [ ] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
M122 Report
```

### Please upload the content of your config.g file.

```gcode
Config.g
```

### Please upload the content of any other releveant macro files.

```gcode

```

### What happened?

Steps to reproduce
```
echo move.axes[0].drivers
```

[image]

Expected result
```
{0.4}
```

Observed result
```
unsupported type code 16
```

dc42, 2025-03-28:
This issue is present in SBC mode only. In standalone mode the response is as expected.

chrishamm, 2025-03-29:
My proposed fix looks like this: